cnlohr/cnixxi #7: Love the concept - request for variation

LouDou, 2023-05-06:
Hi,

I saw this pop up on my YT feed - very interesting and love the concept of an addressable PWM driver for nixie tubes.

I actually have a small stash (now 14; I had 20 in total, but already built a clock using an off-the-shelf board which requires 6) of IN-17 tubes which are a bit smaller (and IMO way cuter :wink:) than the IN-12 you've used.

[image]

[image]

So, could the design be altered to suit the footprint of the IN-17? And is it electrically compatible?

Regards,
Doug

cnlohr, 2023-05-07:
Oh wow! Yes. It should be pretty portable. http://www.tube-tester.com/sites/nixie/dat_arch/IN-17_02.pdf

I also picked up a bunch of IN-4's.  I might poke around to see if I can figure something neat out.

isparkes, 2023-06-09:
IN-4s have a bad reputation for failing - they are not mercury doped and form whiskers between the digits after a while. IN-1 have the same problem. Not saying don't use them, just be aware that you should provide a way of easily changing the tubes , e.g. by using 1mm machine pins. 

A possible optimisation: I have used BSS131 in the past for the cathode drivers - they are a bit cheaper and work just fine. I used them in this proof of concept back in 2018 https://bitbucket.org/isparkes/nixiemodulei2c/ - you'll notice that I used a higher cost controller back then, and managed to squeeze a NeoPixel into the design as well... :D

isparkes, 2023-06-09:
Oh, forgot to say what I actually wanted to say: Love what you've done with that controller! If there's anything I can help out with on with the Nixie specific stuff, feel free to ask. I've had a Nixie obsesession since 2018 or so and know my way around pretty well by now.

ImNoahDev, 2025-05-16:
I see its been a while but did you work this out? I was looking at making a variation for in13 tubes.