cnlohr/cnixxi #10: wht is the float division trick called?

recallmenot, 2023-08-27:
Hi,
this is not an issue:

I was admiring your flyback driver code and I'd just like to know what the trick you used to divide by a float with only integers here is called:
https://github.com/cnlohr/cnixxi/blob/d3db1aef0dade1c4579eeff3b5d0570b02ada58f/firmware/nixitest1/nixitest1.c#L239
Is this something along the lines of continued logartihms?
Are the terms trial and error or is there a trick to finding them?
Sorry if the questions are stupid, I really don't know.

cnlohr, 2023-09-09:
I literally don't know if there's a name for this.  It was something I saw a compiler do a looong time ago in a weird situation and I was like "huh, that's a neat trick."  I don't even recall what compiler it was.

recallmenot, 2023-09-10:
@cnlohr, so I've looked at the code again today, and *now* it clicked:
You're dividing by just-not-enough to cross the target, and then successively approximating the ratio until you're close enough, subtracting ever smaller fractions of the same number, choosing the subtractor to be just a tiny bit smaller than the remaining difference.

The same is also possible from the other direction: divide by just-a-bit-too-much and then add back ever smaller fractions.

And the divisions are dictated by bit-shifts, chosen by trial and error / log(x,2) and then directional rounding?

Sadly I couldn't find a name for this either but I think it's brilliant!
It seems to be related to [division by a constant](https://en.wikipedia.org/wiki/Division_algorithm#Division_by_a_constant), but most of the time, wikipedia is difficult for me to understand.

@emeb
I've looked at the CSD article you linked but couldn't make much of it, but later found Horners method, both and their combination explained exceedingly well in a [TI App note](https://www.ti.com/lit/an/slaa329a/slaa329a.pdf).
I think it's not exactly the same as horner + CSD allows more "quick" precision by using `+` and `-`.

I've exercised them for the 4.44 constant used in cnixxi:
```
binary *< 1
x / 4.44
= x * (1/ 4.44)
= x * 0.224224
= x * 0b0.0011100110101000

CSD
 0 0 1 1 1 0 0 1 1 0 1 0 1 0 0 0
 0 0 1 1 1 0 1 0-1 0 1 0 1 0 0 0
 0 1 0 0-1 0 1 0-1 0 1 0 1 0 0 0
distances
  2     3   2   2   2   2       

horner
xtemp = x >> 2 + x
xtemp = xtemp >> 2 - x
xtemp = xtemp >> 2 + x
xtemp = xtemp >> 2 - x
xtemp = xtemp >> 3 + x
result = xtemp >> 2
```
Does that look right to you? At least my calculator agrees :D

recallmenot, 2023-09-10:
Going to leave this here:
https://github.com/recallmenot/bitshift_div
There's a calculator in there to determine the bit shifts used by @cnlohr s method.

cnlohr, 2023-10-02:
In the future, may want to consider making it a html+javscript thing, so it could be on a github.io page?

But, that's an interesting trick to use + and -.  I have definitely done it with + and -, myself in the past.

cnlohr, 2023-10-02:
I'm adding this to my "to make a video list" but who knows if that will ever happen.