cnlohr/cnixxi #11: KiCad version

vzreagle, 2024-01-25:
Hi, I'm using KiCad 5.1.5 on Ubuntu 20.04 and can't open the schematics. Which version you are using? Thanks, nice project!

closed 2024-02-01 by cnlohr

vzreagle, 2024-02-14:
Yes, I can use the PDF and make my own design, because I want to use only your high voltage generator for another function. But I want to know exactly what transistor you used, because I looked for one with similar characteristics (MOSFET N-CHANNEL 200V 1A) on Mouser and Digikey, and they are very expensive and with packages that do not work for me. Did you buy it at LCSC? could you provide the link? Thanks

cnlohr, 2024-02-15:
You can look into the production files here https://github.com/cnlohr/cnixxi/blob/master/nixitest1/nixitest1-revB.zip

The inner CSV has the BOM.

[nixitest1.csv](https://github.com/cnlohr/cnixxi/files/14289801/nixitest1.csv)

I ended up going with a C2938374 for the high-power switch and C845589 for the individual segments.