cnlohr/cnixxi #13: Add USB control support over rv003usb

JustusFluegel, 2024-04-02:
Hi there, just came across your videos on youtube.

Since you released the `rv0003usb` library after this project was "complete" I thought it might make sense to use that here, if possible (note that I have not looked at the code thus far and I am not sure if for example the pid control loop and usb interrupt could interfere with each others timings in some way) and since you mentioned you wish to use this via usb in the video I just thought to at least throw the idea in the room.

Feel free to just ignore/close this if there is no interest in that, just a thing that  came across my mind :) ( and certainly don't feel pressured to make any changes here, if I wish to use this I can work on this myself (and create a pr afterwards if desired), again, this is just an idea that crossed my mind )

cnlohr, 2024-04-03:
To clarify, this project doesn't use usb at all, it uses a programmer to talk over the SWIO interface to alter the control loop, etc.

It would not be feasible to drive the nixi tube and also be a USB slave at the same time, it would cause serious timing issues in the feedback loop for the boost supply.

Sorry if that quashes any dreams.

closed 2024-04-03 by cnlohr

JustusFluegel, 2024-04-03:
Fair enough, was kinda a long shot anyways (to clarify: I was aware that this doesn't use usb in it's current state, just was interested if it would be theoretically possible to add). 

Thanks for taking your time to answer tho!